afkiwers/inventree_kicad #20: Question: Parameters usage

wolflu05, 2023-09-26:
Hello, this is actually a very cool plugin, awesome work.

While reading the readme carefully, i asked me two things regarding parameters:
1. how should handle the parameters thing. Should I really create parameter templates like kicad_value, kicad_footprint, .. because resistors have their value stored in a parameter template called resistance, capacitors in a parameter template capacitance, ... 
2. how can I define custom footprints for custom components. Is it possible to have them stored in InvenTree too?

So generally some usage examples would be helpful to improve user experience. E.g. I asked me, how do I setup kicad libraries with resistors, there are already a bunch of symbols and foodprints in there, but how do I get them linked to InvenTree? Each resistor may have a different footprint.

afkiwers, 2023-09-26:
Hi @wolflu05,

Thanks for your comments. They are really appreciated!

> how should handle the parameters thing. Should I really create parameter templates like kicad_value, kicad_footprint, .. because resistors have their value stored in a parameter template called resistance, capacitors in a parameter template capacitance, ...

Let me get back to you on this one. I might have an idea how to make this more universal. 
@SchrodingersGat What do you think about one-to-many rel here? It could look for either of them and use first() occurrence.

> how can I define custom footprints for custom components. Is it possible to have them stored in InvenTree too?

HTTP libraries are metadata libraries only. This means you will need matching symbol on footprint libraries. At the moment you would have to manually add it. However, I am already looking at pushing data from KiCad to Inventree. Idea here is: you select a Resistor or so from InvenTree and edit it's properties such footprints, datasheet etc. Then you just hit update or so and it will magically send it to InvenTree.

> So generally some usage examples would be helpful to improve user experience. E.g. I asked me, how do I setup kicad libraries with resistors, there are already a bunch of symbols and footprints in there, but how do I get them linked to InvenTree? Each resistor may have a different footprint.

As for now, this would be done manually. I will add some notes to the readme file.

SchrodingersGat, 2023-09-26:
With regards to the "value" field - what if we add a field to the "KiCad Category" table called "value parameter" - which is a link to a specific parameter to use for the "value" field for that category.

For example, you have a category "capacitors", and the "value" parameter is "Capacitance".

So, for a particular part, the order of priority for generating the "value" is:

1. Look for a parameter value field from the KiCad Catagory table (new field)
2. Look at the global "value field" (this is a setting against the plugin itself)
3. Finally, fall back to the part name

afkiwers, 2023-09-27:
@wolflu05 there is now a variable you can set inside the admin. This will allow you set a default template such as Inductance for that category.

[image]

afkiwers, 2023-09-27:
Addressed in: https://github.com/afkiwers/inventree_kicad/pull/23

closed 2023-09-27 by afkiwers

wolflu05, 2023-09-27:
Thanks for that quick reply and implemenration. That is a very good idea to link them to kicad categories. 

Will it be possible from the kicad side sometime to use foodprints provided by a http lib then? Or do I need to put custom libraries onto a NAS and connect them on every pc kicad is installed on?

afkiwers, 2023-09-27:
Not with the current HTTP lib implementation; that would be something entirely different, unfortunately. The way I implemented the HTTP lib in KiCad really only supports metadata and relies on physical symbol and footprint libs.

What I would recommend is to add a global library of some sort to KiCad and using git or similar as source control. This way you can share libs very easily.

wolflu05, 2023-09-27:
Ok. Thanks for clarification. Makes sense now.