afkiwers/inventree_kicad #25: Error when assigning Parameters

markusEmm123, 2023-09-27:
When i try to assign an Inventree-Category to a KiCAD-Category in the Plugin-Configuration, i get the error message
[image]
I don't have the english version, but the error message should be something like
"No response
No response from Inventree Server
URL: /api/plugins/settings/kicad-library-plugin/KICAD_FOOTPRINT_PARAMETER"

When i click a second or third (or more) time, the input window opens.
I can select a category there, but when i click submit, i get the same message again.

It worked once after some tries for "Symbol".
If i assign a category via admin interface, i can see the corresponding category in the plugin configuration screen.

Please tell me what data elese you need to reproduce this error.

I have Inventree running in Docker with nginx on a a public subdomain.

These are the version information:
# Version Information:
InvenTree-Version: 0.12.7
Django Version: 3.2.19
Commit Hash: 62f44d0
Commit Date: 25/08/2022

Database: mysql
Debug-Mode: False
Deployed using Docker: True
Platform: Linux-4.4.180+-x86_64-with-glibc2.36
Installer: DOC

Active plugins: [{'name': 'InvenTreeBarcode', 'slug': 'inventreebarcode', 'version': '2.0.0'}, {'name': 'InvenTreeCoreNotificationsPlugin', 'slug': 'inventreecorenotificationsplugin', 'version': '1.0.0'}, {'name': 'KiCadLibraryPlugin', 'slug': 'kicad-library-plugin', 'version': '1.1.8'}]

afkiwers, 2023-09-27:
@markusEmm123 no worries my German is pretty good. :-)

It sound like it is timing out for some yet unknown reason. I have an almost identical setup running.
# Version Information:
InvenTree-Version: 0.12.7
Django Version: 3.2.19
Commit Hash: 6563b4c
Commit Date: 2023-09-16

Database: mysql
Debug-Mode: False
Deployed using Docker: True
Platform: Linux-4.4.302+-x86_64-with-glibc2.36
Installer: DOC

Active plugins: [{'name': 'InvenTreeBarcode', 'slug': 'inventreebarcode', 'version': '2.0.0'}, {'name': 'InvenTreeCoreNotificationsPlugin', 'slug': 'inventreecorenotificationsplugin', 'version': '1.0.0'}, {'name': 'KiCadLibraryPlugin', 'slug': 'kicad-library-plugin', 'version': '1.2.0'}]

Can you send through  screen shot of your Plugin Settings page? Also, please update to the latest version ofthe plugin, just to make sure we are not missing anything.

markusEmm123, 2023-09-27:
Thank you for the fast response.
Does not happen anymore with 1.2.0

In parallel i had some problems with Inventree not starting: Worker Threads being killed with Signal 9. This seems to be a hint to missing resources. I have reduced the worker number to 2, now everything is fine again.
Maybe this also had something to do with this issue (timeout).

closed 2023-09-27 by markusEmm123

afkiwers, 2023-09-27:
Awesome !

@SchrodingersGat I had a similar issues with my docker install and had to up my gunicorn timeout a fair bit. Any idea why this might happen? Not enough resources allocated to the containers maybe?

SchrodingersGat, 2023-09-29:
This is often associated with container resource constraints