afkiwers/inventree_kicad #27: INVENTREE_KICAD section is not visible in admin section

markusEmm123, 2023-09-27:
I have activated all plugin integrations in the plugin configuration screen.
I have run "invoke migrate" a few times.
I have restarted the server a few times.

The section is still nowhere to be found in the admin section.

What else can i check/do?

System Information:
# Version Information:
InvenTree-Version: 0.12.7
Django Version: 3.2.19
Commit Hash: 62f44d0
Commit Date: 25/08/2022

Database: mysql
Debug-Mode: False
Deployed using Docker: True
Platform: Linux-4.4.180+-x86_64-with-glibc2.36
Installer: DOC

Active plugins: [{'name': 'InvenTreeBarcode', 'slug': 'inventreebarcode', 'version': '2.0.0'}, {'name': 'InvenTreeCoreNotificationsPlugin', 'slug': 'inventreecorenotificationsplugin', 'version': '1.0.0'}, {'name': 'KiCadLibraryPlugin', 'slug': 'kicad-library-plugin', 'version': '1.2.0'}]

afkiwers, 2023-09-27:
Just to make sure. You did the following?
- docker-compose down
- docker-compose run inventree-server invoke update

What's the console output when it does the migration?

afkiwers, 2023-09-27:
This was my output after invoke update with version 1.2.0

```
Backing up InvenTree database...
Backing Up Database: inventree_prod
Writing file to default-b373eae77842-2023-09-28-024655.dump.gz
Backing up InvenTree media files...
Writing file to b373eae77842-2023-09-28-024711.tar.gz
Running InvenTree database migrations...
========================================
  Running command git clone --filter=blob:none --quiet https://github.com/afkiwers/inventree-kicad-plugin /tmp/pip-install-80xqtunx/inventree-kicad-plugin_cfec37da7f074f90b702980e93d8536a
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Migrations for 'inventree_kicad':
  /usr/local/lib/python3.9/site-packages/inventree_kicad/migrations/0005_selectedcategory_default_value_parameter_template.py
    - Add field default_value_parameter_template to selectedcategory
  Running command git clone --filter=blob:none --quiet https://github.com/afkiwers/inventree-kicad-plugin /tmp/pip-install-jlo7gevt/inventree-kicad-plugin_25c6f6c6e88c49f4afc8d7d0fef9a358
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Operations to perform:
  Apply all migrations: InvenTree, account, admin, auth, authtoken, build, common, company, contenttypes, django_q, error_report, exchange, flags, inventree_kicad, label, order, otp_static, otp_totp, part, plugin, report, sites, socialaccount, stock, taggit, user_sessions, users
Running migrations:
  Applying inventree_kicad.0005_selectedcategory_default_value_parameter_template... OK
  Running command git clone --filter=blob:none --quiet https://github.com/afkiwers/inventree-kicad-plugin /tmp/pip-install-iijtfzfv/inventree-kicad-plugin_a8dba2c5b9cf4385a2e21dbbb0d11909
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Operations to perform:
  Synchronize unmigrated apps: allauth, allauth_2fa, corsheaders, crispy_forms, dbbackup, dj_rest_auth, django_cleanup, django_filters, django_ical, django_otp, djmoney, drf_spectacular, formtools, generic, import_export, maintenance_mode, markdownify, messages, mptt, registration, rest_framework, staticfiles
  Apply all migrations: InvenTree, account, admin, auth, authtoken, build, common, company, contenttypes, django_q, error_report, exchange, flags, inventree_kicad, label, order, otp_static, otp_totp, part, plugin, report, sites, socialaccount, stock, taggit, user_sessions, users
Synchronizing apps without migrations:
  Creating tables...
    Running deferred SQL...
Running migrations:
  No migrations to apply.
```

afkiwers, 2023-09-28:
Can you post a screenshot of your general plugin settings?

markusEmm123, 2023-09-28:
> Can you post a screenshot of your general plugin settings?

If you mean this:
[image]
Then yes, i can ;)

Thanks for posting your "invoke update" output.
Mine is obviously missing the inventree_kicad-parts.

I installed the Plugin by pasting the (pypi-)name to the plugins.txt file and then running update.
Seems like you have installed it from Git (with the installer?).
I will try to uninstall the plugin and then install via Plugin-install-menu via Git.

markusEmm123, 2023-09-28:
After a lot of installing, uninstalling and re-installing, migrating, updating, messing with the database (django_migrations) and manually running migrations, i still don't see the INVENTREE_KICAD-Section in the admin interface.

This is the output when i "invoke update":

```
Backing up InvenTree database...
Backing Up Database: inventree
Writing file to default-inventree-update-db-2023-09-28-202856.dump.gz
Backing up InvenTree media files...
Writing file to inventree-update-db-2023-09-28-202906.tar.gz
Running InvenTree database migrations...
========================================
  Running command git clone --filter=blob:none --quiet https://github.com/afkiwers/inventree-kicad-plugin /tmp/pip-install-0qz8k023/inventree-kicad-plugin_bc86d9e6196e46e1900d238649433bc7
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
No changes detected
  Running command git clone --filter=blob:none --quiet https://github.com/afkiwers/inventree-kicad-plugin /tmp/pip-install-heofxyie/inventree-kicad-plugin_74c426eec0da4c3da14bcc49d638668b
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Operations to perform:
  Apply all migrations: InvenTree, account, admin, auth, authtoken, build, common, company, contenttypes, django_q, error_report, exchange, flags, label, order, otp_static, otp_totp, part, plugin, report, sites, socialaccount, stock, taggit, user_sessions, users
Running migrations:
  No migrations to apply.
  Running command git clone --filter=blob:none --quiet https://github.com/afkiwers/inventree-kicad-plugin /tmp/pip-install-_7woqyui/inventree-kicad-plugin_961a80fdfaf048f5b49ea4a1a16803b7
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Operations to perform:
  Synchronize unmigrated apps: allauth, allauth_2fa, corsheaders, crispy_forms, dbbackup, dj_rest_auth, django_cleanup, django_filters, django_ical, django_otp, djmoney, drf_spectacular, formtools, generic, import_export, maintenance_mode, markdownify, messages, mptt, registration, rest_framework, staticfiles
  Apply all migrations: InvenTree, account, admin, auth, authtoken, build, common, company, contenttypes, django_q, error_report, exchange, flags, label, order, otp_static, otp_totp, part, plugin, report, sites, socialaccount, stock, taggit, user_sessions, users
Synchronizing apps without migrations:
  Creating tables...
    Running deferred SQL...
Running migrations:
  No migrations to apply.
System check identified no issues (1 silenced).
========================================
InvenTree database migrations completed!
```

It seems to be missing inventree_kicad.
I have installed the plugin via pip, plugin.txt file and plugin installation menu, have updated and migrated and restarted a lot of times and have no idea anymore what else to try.

The plugin itself is working very well. To test it I have manually created a database entry to the inventree_kicad_selectedcategory-table and KiCAD can connect and correctly read from the API.
So the database table itself seems to have been created correctly and working correctly.

Does this information somehow help to narrow down the problem?

afkiwers, 2023-09-29:
I seems that the plugin didn't get installed correctly which might be the reason why InvenTree doesn't apply any migrations. The update part probably just doesn't know it's there.

What I did on my docker setup was:
[Through GitHub](https://github.com/afkiwers/inventree_kicad#through-github)
Navigate to the Plugin Settings and click on the Install Plugin button. This will trigger a new window to appear, prompting you to enter the following information:

Package Name: inventree-kicad-plugin
Source URL: git+https://github.com/afkiwers/inventree-kicad-plugin

Then, after it was installed.
1. I restarted the server
2. then I enabled the plugin
3. After enabling I stop InvenTree with docker-compose down
4. Then, invoke update (It should find it here)
5. Then docker-compose ....... up -d

When uninstalling the plugin or trying to reinstall it. Try to also delete the plugin in Plugin Configurations
[image]

Hopefully that helps.

afkiwers, 2023-09-30:
One more thing: make sure you run migrations with the right user; that could cause problems otherwise I believe.

closed 2023-10-03 by markusEmm123

matmair, 2023-10-03:
@markusEmm123 That is quite an important information and probably explains the problems: the plugin.txt file alone is not enough. The plugins need to be installed in the python environment. There is an option to do this on startup but that needs to be enabled in the plugin settings. After enabling you need to run the update command twice - on the second run it should detect the new plugins.
Running two instances or more of the server can be problematic without proper config.

markusEmm123, 2023-10-03:
> There is an option to do this on startup but that needs to be enabled in the plugin settings. 

Thank you for the follow-up. That sounds like a reasonable explanation.
I will consider that for the next update (and run invoke update from the main container).

Unfortunately Synology Docker does not offer an option to change the startup command after the container has been created. Therefore updating the database structure from a second container is the only option I see at the moment.
But that should go to the Synology support and not here...

SchrodingersGat, 2023-10-03:
@markusEmm123 this is a setting you can set within InvenTree itself:

[image]

afkiwers, 2023-10-03:
@markusEmm123 it seems we have pretty much the same setup running. I run my InvenTree containers on a DS1819+.

I'd like to add a few things here, in case someone else runs into similar issues:

You can open a terminal and do the migration from within if you wanted. However, I didn't actually do that but below a quick example of how that would work.

Navigating to the Container Manager and double-clicking on your running inventree container, you can open a terminal; just hit the Action button -> Open Terminal.
[image]

When in there simply execute: **python manage.py migrate**.
That should do it.

[image]

However, if you use the docker-compose file and a dedicated folder for InvenTree you could also SSH into your synology. 
Using sudo su go to your inventree folder where the docker-compose file is located. In my case it would be cs **/volume1/docker/inventree**

Inside your InvenTree folder execute **docker-compose run inventree-server invoke update**. You don't even need to **docker-compose down** the current instance.

[image]