afkiwers/inventree_kicad #40: Default Symbol does not work

30350n, 2023-11-16:
Setting the Default Symbol for a category to `Device:R` does not work for me. I think this was broken by #39.
[image]
[image]

afkiwers, 2023-11-17:
@30350n I created a fix for that. 

 Please update to version 1.3.2 and let me know if that resolves your issue.

30350n, 2023-11-17:
Will check once I'm done investigating the cause for the other issue I openend :)

closed 2023-11-17 by afkiwers

afkiwers, 2023-11-17:
It does if you run an earlier version but the latest should definitely support it.

It is working on my end
[image]

30350n, 2023-11-17:
Yep, works, thanks for the quick fix! :smiley: