afkiwers/inventree_kicad #41: Part URL is not fully qualified

30350n, 2023-11-16:
When creating a symbol in KiCad, the Part URL is not fully qualified.
[image]
[image]

30350n, 2023-11-17:
(Currently investigating this, should hopefully be able to fix it myself ^^)

afkiwers, 2023-11-17:
@30350n  what inventree version are you running? Would you be able past the full log?

30350n, 2023-11-17:
Master from a few commits ago, shouldn't really matter though.
The core issue seems to be that I simply don't have a `INVENTREE_BASE_URL` set. I'm wondering why [this](https://github.com/afkiwers/inventree_kicad/blob/main/inventree_kicad/serializers.py#L246) fallback isn't working though ...

30350n, 2023-11-17:
Aka: Why is there no request there?

PR #43 (2023-11-18): Make part url fully qualified even if INVENTREE_BASE_URL is not set ( Fix #41 )
This fixes #41.
closed 2023-11-18 by afkiwers