afkiwers/inventree_kicad #46: [HOW]Divide http libraries by category inside kicad

Nudelsalad, 2023-12-05:
I implemented the http library feature for my Inventree instance and was wondering if it would be possible to split all the different categories (Caps, Resistors, etc) into different libraries for a "better overview". I tested the current implementation and also tried modifying the root url according to: [KICAD_API_LINK](https://dev-docs.kicad.org/en/apis-and-binding/http-libraries/) (section Example) with e.g.:

https://inventree.example.de/plugin/kicad-library-plugin/parts/category/31.json 

where 31 is the category for all capacitors. But it still showed all parts of all categories.

Am I missing sth or is this not possible so far? According to the Kicad-API it should be. 
Appreciate any help!

afkiwers, 2023-12-05:
@Nudelsalad did you add categories in the backend?

Without any, it will show all parts, see code snipped below
[image]

[image]

afkiwers, 2023-12-06:
The config file does not contain any information about the categories; it merely tells KiCad where to find the library and what user token to use.

To add categories you have to head to the admin panel and open the KiCad model. Please have a look at section 'Adding Categories to KiCad' which you can find in the readme file.

If you don't have that model in your admin panel you might be missing the migration step.

Nudelsalad, 2023-12-06:
I am sry, maybe I expressed my problem wrongly:
I was able to create the categories inside inentree, added 4 Parts as example(Cap, Resistor, XTal and Zener) and created in the admin panel the corresponding kicad categories:
[image]

There are 4 parts of differnet category inside my instance:
[image]

My Problem now is that these 4 categories of different component categories (Caps, Resistors, etc.) are all mixed in one kicad library which is not very nice but if it is not implemented so far i guess i have to go with it
[image]

I understood that the rooturl so far is only a dumb url pointing to the api but is it possible to spread this into different libs or how are y'all working with it? Having hundreds of components inside one lib seems to be a mess

afkiwers, 2023-12-06:
KiCad should show you 7 different libs inside the symbol picker. Each selected category represents one library. I'll send some screenshots later when I'm in the office.

Would you be able to send through your InvenTree version data and a screenshot of the plug-in settings?

This behaviour is definitely not expected.

Nudelsalad, 2023-12-06:
So it is a clean fresh docker container install of the latest version of inventree, in the plugin section of the kicad-library-endpoint is everything on default. In my general plugin section i have this:

[image]

I installed the plugin via pip inside inventree like mentioned in the readme.
I uninstalled the plugin, removed all categories and build everything up again and I have still everything in one lib in kicad
The http lib is under the global libs inside kicad. I also pulled the latest nightly and reinstalled it. But in my symbol picker the categories still appear under one lib 

[image]

WOW
I just figured out that the symbol picker inside the schematic is different from the one you can access from project menu....

[image]

I am so sorry for bothering you, I have never used this feature before very thanks for your support

I have still one question left
I understood how to work with the httplib feature and symbols inside inventree, but how to work efficiently with footprints? How did you assign footprints to your parts? Can i do this on part creation? with a separate field? or everything only one by one from admin panel? FYI: I dont want to create separate footprint subcategories inside every category to assign the footprints. Is there an efficient way to do this?
I will close the issue upon your answer Thanks!

closed 2023-12-06 by Nudelsalad