afkiwers/inventree_kicad #53: BOM Import

SchrodingersGat, 2023-12-07:
It would be handy now that this plugin has an XML importer to be able to directly import a BOM

afkiwers, 2023-12-07:
I guess a simple select field somewhere could be used which contains all assemblies and is filterable. If something is selected, it would create a BOM. But what if there is already a bom?

[image]

SchrodingersGat, 2023-12-07:
> But what if there is already a bom?

As a starting point, could raise an exception if BOMItems exist, and the user has to clear out the BOM manually before uploading?

tnemrap, 2023-12-09:
I am also interested of a function to create/update an bom on an assembly.

afkiwers, 2023-12-09:
This feature is already available using InvenTree's BOM features.

I have been thinking about this the last couple of days and can't really see the benefits of having another way to create BOMs as importing BOMs using the BOM import tools from InvenTree are really working well; at least for me.

This being said, if anyone likes to make a pull request, go for it. 😉

closed 2023-12-14 by afkiwers