afkiwers/inventree_kicad #57: Can not open InvenTree.kicad_httplib in KiCad

tnemrap, 2023-12-08:
I know that this should be in KiCAD but i try here when you have done the Pull in Kicad.
I can not open the created InvenTree.kicad_httplib from Kicad 7.09 Symbol Library, it is not an selectable extension.
[image]

30350n, 2023-12-08:
HTTP Libraries are only supported in nightly (7.99) builds.

afkiwers, 2023-12-08:
Let's hope they stick to their release schedule and release 8.0 next month.

tnemrap, 2023-12-08:
Is nightly stable enough to run?

30350n, 2023-12-08:
Long Answer: https://www.kicad.org/help/nightlies-and-rcs/
Short Answer: No.

afkiwers, 2023-12-08:
I haven't had any issues with most of them. However KiCad devs won't guarantee 'data loss'; whatever that would look like.

We already use the 7.99 in production for smaller projects.

tnemrap, 2023-12-08:
What build are you using, i will not hold you responsible. :)

afkiwers, 2023-12-08:
I just updated my version recently.

Application: KiCad x64 on x64

Version: 7.99.0-3795-g4268ffec77, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.12.1
	HarfBuzz 8.2.1
	FontConfig 2.14.2
	libcurl/8.4.0-DEV Schannel zlib/1.3

Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec  3 2023 06:17:56
	wxWidgets: 3.2.4 (wchar_t,wx containers)
	Boost: 1.83.0
	OCC: 7.7.1
	Curl: 8.4.0-DEV
	ngspice: 41
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:

closed 2023-12-08 by afkiwers

tnemrap, 2023-12-08:
Worked like a charm.