afkiwers/inventree_kicad #59: Allow part selection during metadata import rather than discard part

atanisoft, 2023-12-14:
Currently when importing an XML from KiCad if the component does not have a field name starting with "inventree" the component is discarded and not processed. It would be great if we could manually specify the mapping during import for existing PCBs that do not currently have the field data defined. It would also be great if an import summary could be generated after import listing components that were imported (grouped by part) and which could not be imported due to no part match or missing field (grouped by part / ref)

afkiwers, 2023-12-14:
I'll split this into two separate issues.

Adding a generic identifier for the part ID is quickly done. However, creating a table etc will  probably take a couple of hours.

afkiwers, 2023-12-15:
@atanisoft I implemented the id mapping and created a new release.

atanisoft, 2023-12-15:
Thanks @afkiwers I pulled in the new release to my docker based instance, changed the key to "mpn" (as a temporary option) and was going to import the XML file. Is there an easy way to follow the process to see that it imported things correctly?

afkiwers, 2023-12-15:
No, not at the moment, sorry.

atanisoft, 2023-12-15:
might need a small follow up PR for the non-numeric value in a key to lookup by part name rather than only part id. But that is minor. I can confirm that it works if I manually add a numeric only key value.

afkiwers, 2023-12-15:
You mean instead of using the InvenTree part ID use the InvenTree name instead?

atanisoft, 2023-12-15:
Yeah, as a fallback for non-numeric or numeric values that don't match as a part-id. As an example, I've got an existing schematic with "mpn" as a key for tracking the manufacturer part number (such as 860010575013, Wurth Electronik 100uF 35V electrolytic cap) which when imported will try to match by part-id and find it doesn't exist but it has an existing part in the system (for this example it's part-id 100)

closed 2025-03-14 by afkiwers