afkiwers/inventree_kicad #70: Featurerequest: Virtual Parameters

TimGoll, 2024-02-01:
I'm not 100% sure if this is the right place to ask. Maybe this should be directed at inventree itself. But I try it here first anyway.

I'm using [this tool](https://github.com/30350n/inventree_part_import) to import LCSC parts into inventree. I'm not sure that this is the best way to do it, but this is how I do it right now.

It works really good, but it adds stuff such as the `Supplier Part Number` to the supplier tab. Which makes sense if you look at it that way. However I'm unable to get this value in KiCad and I'd like to have the `LCSC Part #` as a parameter in KiCad so the exported BOM works to order the PCBs.

The easiest solution for me would be to edit the aforementioned tool. But I'm wondering if you could add to your addon some sort of virtual parameter that adds the LCSC part number (or any MPN for that matter) to the KiCad parameters.

Maybe something in this menu:
[image]

Sorry for the rambling and feel free to close if this is out of scope for this addon!

afkiwers, 2024-02-01:
Could you send through some screenshots of the data you want in kicad?

TimGoll, 2024-02-01:
Mainly this one:

[image]

But it probably needs a different name so that the KiCad JLC Plugin accepts it. [Here's a list of the names that are accepted](https://github.com/bennymeg/JLC-Plugin-for-KiCad/blob/e9f4b5b29b02cef3289655887143cbf6f7f6a05f/plugins/process.py#L287-L288) so it would probably be best to be able to define it in the UI.

Also the name / value are something that I'm not that much of a fan of:
[image]

I set the value to the part value (which is fine, even though I'd like it to be shortened automatically to `2u2` for example; but I can do that by modifying the import tool I guess), but the name is a bit annoying. The name is fine in inventree, in fact I think it is even a good name, but it is confusing in KiCad imho. I'd like it to be something like `100n_0402_LCSC` or so - but without modifying the actual name in InvenTree.

Again, I'm not sure if this is in scope for this addon or if this is even something anyone would want from it. Maybe this is just my weird way of using it. But as a general feedback, this is something that bothers me.

TimGoll, 2024-02-01:
I'd like to also add that the last point is only something I'd want for basic parts such as capacitors. Because I'll have quite a lot of them in the list and it would make it easier to find them in KiCad if their name was less cryptic.

So if this is something you'd like to add, then it is something that should be in the categories settings

afkiwers, 2024-02-01:
The plugin should only provide "non-volatile" data to KiCad to keep the schematic simple and consistent over time. Let's say your supplier is shutting down and you would need a new one, KiCad would still display the old supplier unless updated.

InvenTree got the facilities to cater for that.

TimGoll, 2024-02-02:
So I guess it comes down to me being unfamiliar with InvenTree

afkiwers, 2024-02-02:
We actually work the other way around. We import BOMs from KiCad into InvenTree part assembly BOMs which then shows you what's in stock and what's missing. Then you can create a purchase order for a specific distributor and upload that as csv file almost straight into your cart. Has been working like a treat so far. 

I think there are actually plugins in the work which will automate that even more.

TimGoll, 2024-02-02:
Looking forward to it! This is still all a bit new to me and I have yet to learn what is coming. Is there somewhere a resource for stuff like this? So I can read into it more? Because [this list](https://docs.inventree.org/en/latest/extend/integrate/) seems to be a bit out of date

afkiwers, 2024-02-06:
The InvenTree's docs are quite good I think. It's just a lot to go through.

It helps to have a look at InvenTree's demo server. It's pretty cool what it can do.

afkiwers, 2024-03-02:
@TimGoll just letting you know that I am currently looking into adding this feature.

TimGoll, 2024-05-28:
awesome, I'm looking forward to it

ondras12345, 2024-11-23:
I have a problem that could potentially also be solved by something like virtual parameters / templates.
I have an electrolytic capacitor in InvenTree, and it has separate parameters for capacitance and rated voltage. However, I'd like the value shown in KiCad to be `{capacitance} / {rated_voltage}`.

afkiwers, 2024-12-02:
Hi Tim, Unfortunately I don't really have a lot of time to work on that plugin at the moment. Sorry about that.

TimGoll, 2024-12-02:
> Hi Tim, Unfortunately I don't really have a lot of time to work on that plugin at the moment. Sorry about that.

Understandable. Thank you!

TimGoll, 2024-12-05:
My team manager is open to chip in a few of my dev hours to implement such a feature. Could you line out how you'd do it, so that it matches your ideas. Thanks!

afkiwers, 2024-12-06:
Hi Tim, I am happy for you to lead this one. 

Here is probably the best place to discuss your ideas. In general, as long as it doesn't disturb any of the other features I'm open to whatever helps you.

afkiwers, 2025-03-18:
@TimGoll are you still interested, otherwise I'll close this one for now.

TimGoll, 2025-03-18:
Interested yes, the time is an issue

LqdBcnAtWork, 2025-10-06:
So if I follow correctly, you want to map _`some value`_ (eg, a supplier's part number), to _`some parameter`_ that will get shared with KiCad?

Example, some resistor with IPN of `RT2820200`, is supplied by Digi-Key under the number `13-RSF200JB-73-2R-ND`, you want that number to show up here?

[image]

My only question is _why?_

For a KiCad integration? So that a supplier number could be injected from Inventree into KiCad?

If that's the process you want or have settled on I won't question it further. A working solution is often more important than the best solution.

I just find it a little bizarre. While we don't actually use Inventree or KiCad (we have low priority plans to switch to both _eventually_). Our workflow has ordering parts and PCBs entirely separate from our PCB design software. The manufacturing files are made and versioned by the designer, and the BOM is imported into our inventory control. Then parts are ordered using the information in inventory.

The PCB design software has little to no need to know that `RT2820200` is actually `13-RSF200JB-73-2R-ND` from Digi-Key. Our system is also setup this way because we use multiple PCB manufactures, a separate vendor for pick&place, hand solder a lot of our boards and all of the through hole components, and purchase all components ourselves. No full service board manufacturing for us 🤷.

IMO if you have little to no use for Inventree's ordering capabilities for components (via POs etc), your idea to modify the import tool to move the data into a parameter might be best.