afkiwers/inventree_kicad #76: HTTP "302:Found" with KiCad 8.0rc2

markusEmm123, 2024-02-08:
As the new release candidate of KiCAD natively supports HTTP libs now, i wanted to try the connection to Inventree.
I get an error:
"Can not connect to https://inventree.example.de/plugin/kicad-library-plugin/v1/ 
API has responded with error code: 302: Found".
I'm using Inventree 13.5 and the plugin version 1.3.14 with KiCAD 8.0.0rc2.
I'm using a https connection to a public URL.
My library-file looks like this:
```
{
    "meta": {
        "version": 1.0
    },
    "name": "Inventree HTTP Library",
    "description": "KiCad library sourced from Inventree REST API",
    "source": {
        "type": "REST_API",
        "api_version": "v1",
        "root_url": "https://inventree.example.de/plugin/kicad-library-plugin/",
        "token": "<mytoken>"
    }
}
```

If I open the URL in the browser, i get something which I assume is valid data.
`{"categories":"http://inventree.example.de/plugin/kicad-library-plugin/v1/categories/","parts":"http://inventree.example.de/plugin/kicad-library-plugin/v1/parts/"}`
as plain text in the browser.

If i leave out the trailing slash though (as shown in the documentation on this github page), (https://inventree.example.de/plugin/kicad-library-plugin) I get redirected to the main inventree side.

Any ideas how to solve this or on what level (Inventree, this plugin or KiCAD) the problem could be located?

afkiwers, 2024-02-08:
Would you be able to provide access to the library? You can send it to iwers11@gmail.com and I have a quick look.

Your version and the settings file look fine.

markusEmm123, 2024-02-08:
While generating account data for you I realized I'm still using an old token.
Inventree changed that sometime in the meantime.
With the new (now right) token everything works fine.
The error message is a bit distracting though.

closed 2024-02-08 by markusEmm123

afkiwers, 2024-02-08:
That's good!

I will have a look if I can tweak those KiCad messages.