afkiwers/inventree_kicad #83: build_absolute_uri with HTTPS

DesertLion1, 2024-03-02:
Hello to everyone,

my InvenTree installation is running behind an nginx reverse proxy with https. The X-Forwarded-Proto header is configured and https works properly throughout InvenTree. However, the absolute URIs inventree_kicad generates do not seem to respect this setting and are all only http://

For instance, the request to **https://my-inventree.example/plugin/kicad-library-plugin/v1/** returns
{"categories":"http://my-inventree.example/plugin/kicad-library-plugin/v1/categories/","parts":"http://my-inventree.example/plugin/kicad-library-plugin/v1/parts/"}

Any ideas what is missing? 

inventree_kicad: 1.3.16
InvenTree: 0.13.5
Django: 3.2.23

Regards
Stefan

afkiwers, 2024-03-18:
@DesertLion1 I assume you fixed the problem? I will close this issue for now as we have heard back from you.

Feel free to reopen it at any time.

closed 2024-03-18 by afkiwers

Kamaar, 2024-12-29:
Hi, sorry to bother but I'm having the same problem.
My istance of inventree is on https://inventree.local and is working correctly.
When I try the link on the plugin interface I get:
`{
"categories": "http://inventree.local/plugin/kicad-library-plugin/v1/categories/",
"parts": "http://inventree.local/plugin/kicad-library-plugin/v1/parts/"
}`
the env INVENTREE_SITE_URL is set correctly:
`INVENTREE_SITE_URL=https://inventree.local`

Also in kicad is giving :
`Errors loading symbols:
Error loading symbol library #Inventree.
Could not connect to https://inventree.local/plugin/kicad-library-plugin/v1/. Errors: API responded with error code: 0: Unknown`

What I can check/verify? is a couple of day that I'm trying to resolve it.

Thanks

SchrodingersGat, 2024-12-30:
@Kamaar have a look at the [server access](https://docs.inventree.org/en/stable/start/config/#server-access) documentation.

In particular, you should set `INVENTREE_USE_X_FORWARDED_HOST` to `True` - either in your .env file on via your configuration file.

It looks like the `build_absolute_uri` method is using the `http` header rather than `https` header.

Kamaar, 2024-12-30:
Thanks for the tip.
I've put the INVENTREE_USE_X_FORWARDED_HOST=True in the .env, but unfortunately I don't see any changes, rebooted and updated the docker containers. (also true in the config.yaml file)

SchrodingersGat, 2024-12-31:
@Kamaar looking into this further, it appears that the current "stable" InvenTree server does not support the `X-Forwarded-Proto` header (it will ignore it)...

So, I have just submitted [this patch](https://github.com/inventree/InvenTree/pull/8797) to address this. I believe that this should be the source of the issue here...

You can wait for the upcoming `0.17.2` release, or try the `inventree:inventree/latest` docker image

Kamaar, 2024-12-31:
Thanks for the quick fix, I'll wait for the upcoming release.

Thanks again and happy new year!

Kamaar, 2025-01-09:
@SchrodingersGat Just to let you know that I've updated to the 0.17.2 but the problem persists.

Thanks

SchrodingersGat, 2025-01-10:
@Kamaar is the internal InvenTree server running at `http` and being reverse proxied from `https`, correct? There may be some subtlety in the `build_absolute_uri` method which is not observing this properly