afkiwers/inventree_kicad #87: migrate database

StecklerCircuits, 2024-03-20:
you say "migrate database" several times in the setup.  please forgive me for not knowing, but could you please elaborate or show a command for that?  
i'm running inventree in a docker container.

SchrodingersGat, 2024-03-20:
https://docs.inventree.org/en/stable/start/docker_install/#update-database

In your case, to just perform database migrations, run:

```
docker compose run --rm inventree-server invoke migrate
```

Or, set the [INVENTREE_AUTO_UPDATE](https://docs.inventree.org/en/stable/start/config/#basic-options) setting to automatically run database migrations!

afkiwers, 2024-04-05:
@StecklerCircuits did you get all the information needed to get it to work?

closed 2024-05-02 by afkiwers