afkiwers/inventree_kicad #96: Question on using the plugin - where is footprint info stored?

StormTrooper, 2024-05-19:
Been using KiCad and Inventree for a while now and just discovered this plugin which looks very useful. 

I've followed the installation document and managed to install the plugin (v1.4.1) with KiCad 8.0.2. 
Setup the API/categories etc and in the symbol library I can see the list of parts. 

So far so good. 

One thing I'm not following is where do you setup the symbol and footprint information for the part. 
I get the Inventree has no way to know what the symbol and footprint will be and this has to be manually entered for each part. 

[image]

Where do I do this though. I don't see anywhere in the part where this information can be entered?

[image]

Am I correct in thinking the footprint info be stored in Inventree?

afkiwers, 2024-05-19:
Hi @StormTrooper,

The plugin looks for parameters. An example below.
[image]

You can define which parameters are used in several places. For a generic assignment you can use the plugin settings. Please note that there is a bug in inventree which means it won't display the name only the ID. If you click on it though it will show you the names.
[image]

If you want to specify a specific value or footprint parameter for a certain category, you can do that inside the admin panel. This will override the generic plugin settings.
[image]

StormTrooper, 2024-05-20:
Thanks for the details. That's the bit I was missing. 

Added the missing parameters to KiCad and then setup the plugin to use those parameters. 

Working perfectly now.

closed 2024-05-20 by StormTrooper

StormTrooper, 2024-05-20:
@afkiwers  I thought I had it all working, but one thing that is confusing me. 

In Inventree I've for a part setup as follows:
[image]

When placing the part in KiCad I get the IPN | Value:

[image]

What do I need to change to just display the value. (100k)
Played around with a few settings but not having any luck.

afkiwers, 2024-05-20:
Did you enable Include IPN in the settings? It looks like it's set to **include and show in schematic **

StormTrooper, 2024-05-20:
[image]

[image]

[image]

afkiwers, 2024-05-20:
InvenTree is providing that to the plugin. Have a look under InvenTree settings->part->'Part Name Display Format' and play with that format. You probably have to get rid of some of the formatting.

Sorry, I don't have access to my computer at the moment just got my phone which makes it a bit harder to quickly check stuff on my side.

afkiwers, 2024-05-20:
What's displayed as part name at the top in InvenTree when opening the parts page?

StormTrooper, 2024-05-20:
Thanks, looks like it's the formatting.  If part.ipn exists then display it:

```{{ part.IPN if part.IPN }}{{ ' | ' if part.IPN }}{{ part.name }}{{ ' | ' if part.revision }}{{ part.revision if part.revision }}```

Same with revision. 

The part name is displayed as ```RES-100K-0805 | 100k``` in Inventree. 
I'll modify the formatting.

StormTrooper, 2024-05-20:
Changed it to ```{{ part.name }}``` which fixes the issue. 

Made a note of the old setting in case something breaks down the line which expects that format.

afkiwers, 2024-05-20:
Fantastic!

That's just a display thing and should not break anything.