afkiwers/inventree_kicad #97: Missing port in the datasheet fields

simoneamadori, 2024-06-20:
In the datasheet file the address is correct but is missing the port like the part URL:
[image]

afkiwers, 2024-08-28:
@simoneamadori sorry for the late reply. Did you resolve it? I just checked and it is displayed correctly in my setup.

"datasheet":{"value":"http://192.168.10.8:1337/media/part_files/264/ACS71240-Datasheet.pdf"

Did you set the base URL properly?
[image]

simoneamadori, 2024-08-28:
Don't worry is not a big problem.
It seems that the problem is there, but unfortunately I cannot change it "Site URL is locked by configuration", I'm running in docker and I've tried to set the INVENTREE_SITE_URL=http://192.168.10.18 with the port ":1337" in the .env file, but then I cannot access the web interface anymore.

afkiwers, 2024-08-28:
Alrighty, I will close this issue then.