afkiwers/inventree_kicad #103: Stock count for template part does not include variants

ondras12345, 2024-11-23:
I have [a connector](https://www.tme.eu/en/details/t821-1-06-s1/idc-connectors/amphenol-communications-solutions/t821106a1s100ceu/) that I need to use two different symbols for.
I tried to solve this by having a part called T821106A1S100CEU with `Connector_Generic:Conn_02x03_Odd_Even`, and a template part called  AVR-ISP-6 with `Connector:AVR-ISP-6` and T821106A1S100CEU added as a variant.
When I open the AVR-ISP-6 part in the InvenTree web interface, it shows available stock: 2 (even though the only stock item is for T821106A1S100CEU). However, when adding a symbol in KiCad, it shows "AVR-ISP-6   [Stock: 0] 6-pin Atmel AVR ISP programming connector".

I also tried querying the same information via the HTTP API, and it also shows as 2:
```python
In [1]: from inventree.api import InvenTreeAPI
   ...: from inventree.part import Part
   ...: from getpass import getpass

In [2]: INVENTREE_URL = getpass()
Password: 

In [3]: INVENTREE_TOKEN = getpass()
Password: 

In [4]: api = InvenTreeAPI(INVENTREE_URL, token=INVENTREE_TOKEN)

In [5]: p = Part.list(api, search="AVR-ISP-6")[0]

In [6]: p.getRequirements()
Out[6]: 
{'available_stock': 2.0,
 'on_order': 0,
 'required_build_order_quantity': 0,
 'allocated_build_order_quantity': 0.0,
 'required_sales_order_quantity': 0,
 'allocated_sales_order_quantity': 0.0,
 'allocated': 0.0,
 'required': 0}
```

Plugin settings:

|                           |   |   |
|:--------------------------|:--|:--|
|Display Available Stock    |When activated, the plugin will provide stock information which will be displayed as part of the top level description in KiCad|On|
|Stock Count Display Format |This will be displayed after the part's description in KiCad (right column). Note: {1} contains the Stock information, {0} the description of the part.|[Stock: {1}] {0}|

afkiwers, 2025-03-14:
@ondras12345 is this still happening?

ondras12345, 2025-03-15:
Still happening on InvenTree 0.17.8 with `inventree_kicad` 1.4.5.

afkiwers, 2025-03-17:
@SchrodingersGat would you be able to have a look at it?

SchrodingersGat, 2025-03-17:
You are using the `annotate_total_stock` filter method which does not include variants.

Check out how the "total including variants" is calculated here:

https://github.com/inventree/InvenTree/blob/9db5205f79c9f88ea9e276546a883be860d3eb5c/src/backend/InvenTree/part/serializers.py#L845

And variant stock total is calculated thusly:

https://github.com/inventree/InvenTree/blob/9db5205f79c9f88ea9e276546a883be860d3eb5c/src/backend/InvenTree/part/serializers.py#L806