afkiwers/inventree_kicad #104: Extra Symbol fields

TimGoll, 2024-12-05:
maybe related to #70 

[image]

In this picture on the left you see how the inventree part looks, on the right is how my manual parts look. The most important difference is that on the right I show the rating, in this case the forward voltage. Your API access already sets this value, but by default it is hidden. Is it possible to make it visible by default? I was unable to find a setting for that, but I might have missed something.

The second thing is probably something you can't do: As you can see on the right, I moved the labels to be tighter together, is it possible to set this as a default? I think not, because KiCad seems to overwrite the text field positions even if defined in the symbol itself.

TimGoll, 2024-12-05:
It seems like it is not possible, right now you set everything to false:

https://github.com/afkiwers/inventree_kicad/blob/c512f1e1102daaabe096c02f3868f69de97ec471/inventree_kicad/serializers.py#L296-L299

I'd love to be able to chose between "visible", "not visible" and "do not include"

afkiwers, 2024-12-23:
Hi Tim, sorry for my late reply. I'm am currently overseas (Germany actually) with limited access to respond/review things.

I'll be back home in January.

In general, I'd be interested to have some sort of setting that lists all the fields available where you can toggle the visibility  so it's displayed in KiCad.

TimGoll, 2024-12-23:
> Germany actually

Greetings, I live in Germany!

Let's talk about this again in January then

PR #108 (2025-03-14): Add an option to make Part Parameters visible by default
**Fixes #104**

One issue my team has found was when we use InvenTree parts on KiCad, we want to have things like Voltage Ratings on capacitors show up, but only for capacitors. 

As of the most recent release from when I made this Pull Request, all user-created parameters in InvenTree were always set to invisible by default with no option of turning them on unless you edit the plugin code. 

My change makes it simple to modify what parameters are visible from a parameter named `Kicad_Visible_Fields`. When combined with a category part p [... cut]
closed 2025-03-14 by afkiwers