afkiwers/inventree_kicad #107: Workflow evaluation questions

gunstr, 2025-03-06:
As InvenTree has stopped supporting BoM import using IPN as matching key as discussed [here](https://github.com/inventree/InvenTree/issues/9248) I'm evaluating alternative workflows and using `inventree_kicad` as a work around has been suggested.

Before spending time to evaluate this integration more in depth I have a couple of questions though. Not really an issue, but as there is no discussion section in the repo I ask them here.

1.  As this plugin sits in the middle between InvenTree and KiCad it has to keep up with potential interface changes in both.  Do you expect that to be manageble?
2. An example of the above as mentioned in the README is that the `metadata import tool` can no longer be used from KiCad8 and onwards. Given that we have a a rather long history with KiCad a kick start to get existing symbols into Inventree is necessary. Are there any other suggestions or ideas/plans for how to resolve this?
3. To avoid too large impact on existing projects I would prefer not to alter our current KiCad library structure or symbol names. The short term goal is just to add the InvenTree "pk" to existing symbols. Reading the documentation it's not clear to me if this is possible or not. What's your view? For any new parts this is of course less of a problem and it looks really nice to manage the KiCad symbol metadata directly in InvenTree 😄

afkiwers, 2025-03-14:
@gunstr 

> As this plugin sits in the middle between InvenTree and KiCad it has to keep up with potential interface changes in both. Do you expect that to be manageble?

I don't know KiCADs roadmap. So, maybe.

> An example of the above as mentioned in the README is that the metadata import tool can no longer be used from KiCad8 and onwards. Given that we have a a rather long history with KiCad a kick start [...]

You could use an old KiCAD version or parse your schmetic files and create a suitable import file. Otherwise, not really. A lot of users have the same problem.

> To avoid too large impact on existing projects I would prefer not to alter our current KiCad library structure or symbol names. The short term goal is just to add the InvenTree "pk" to existing [...]

You can add whatever data you like from InvenTree. There are some default things like the datasheet but everything else is configurable.

We are using this plugin a lot at work and it makes our workflow a lot smoother. There's also more control as to what parts are used and engineers can check quickly from within KiCAD what caps, res etc are available right now.

Also, having the footprints and symbols controlled in a repo eradicates footprint or symbol errors almost entirely.

gunstr, 2025-03-15:
@afkiwers Thanks for your input. It looks promising and I will do a test of the plugin and possible workflows as soon as I can.

closed 2025-03-15 by gunstr