afkiwers/inventree_kicad #113: Feature Req: More parameters for Description Format

LqdBcnAtWork, 2025-07-04:
Currently the only parameters for "Stock Count Display Format" are description with `{0}`, and stock count with `{1}`.

Maybe this should be generalized to allow for more parameters from the part?

Perhaps something like this:

```python
part_ns = SimpleNamespace(
    name=part.name,
    IPN=part.IPN,
    description=part.description,
    stock=stock_count,
    revision=part.revision,
    used_in=0
)

if hasattr(part, "get_used_in"):
    part_ns.used_in = len(part.get_used_in())

description = self.stock_count_format.format(part.description, decimal2string(stock_count), part=part_ns)
```

This would allow us to change the default from `[Stock: {1}] {0}` to `[Stock: {part.stock:.0f}] {part.description}`.

Or in my case: `[Stock: {part.stock:_>5.0f}] [Used In: {part.used_in:0.0f}] {part.description}`

Two main benefits over the current method:
1. Better self-documentation. Currently the setting's description is doing a _lot_ of work to document its use.
2. More functionality and easier to extend. Like the `used_in` param I added.

I (semi obviously) already have this implemented. I suppose we'll start the discussion here, and if desired I'll create a pull request.

afkiwers, 2025-07-04:
I'd be open to that. However, the KiCad devs don't like the fact that the data inside the part picker is different to what ends up in the schematic.

So, unsure if that's worth changing as this little "hack" how they call it will be removed once the new hattp_lib 2.0 comes out.

afkiwers, 2025-07-07:
When I designed the rest api integration for KiCad I had months of discussions until it was accepted and this sneaky feature was rather an unintentional accident.

However, I fully agree with what you are saying about sneaking data in and at that stage. Anyway, it caused me a lot of trouble and apparently upset some people unfortunately.

I'm happy for you to implement that. Only requirement from my side would be, it needs to be backwards compatible to the current format without any additional switches. There are too many already.

Or rewrite the format automatically when the old one was saved before updating.

LqdBcnAtWork, 2025-07-14:
So far my plan is to collapse the amount of different kinds of requests from 4 to 2. Push for `Content-Encoding`, and allow the *server* to dictate a few extra things via HTTP headers. It can send `Last-Modified` and `Refresh` headers, and HTTP `206` codes for deltas.

Sure it can make the server more complicated. But it's also completely optional. If the server sends no `Refresh`, the client won't automatically check again. Deltas are completely optional. Just never send HTTP `206`s.

Maybe I should add one more request type. Just a single part via its `id`, so if the client ever gets confused or decides its data is dirty, it can request a single part. Not sure about that one yet. Might be useful.

Deltas could be calculated if there was a `last_modified_date` field on parts. Maybe it's worth trying to up-stream such a field into inventree? Might be complicated, as changes to Parameters and BoM items might want to touch that date as well. Unless they had their own date fields? Just thoughts at this point.