afkiwers/inventree_kicad #118: Part Parameter Field Visibility settings are broken after version 1.4.6

MrARM, 2025-08-19:
I updated the plugin to the latest version on my Inventree stack and I've found that there's something wrong with the way that the global part parameter variable is being used with the changes brought in #112. One of our engineers noticed that the part parameter template we have is being dragged in as a parameter and it was visible by default.

[image]

The local part parameter option only allows me to select one parameter, and it appears to be storing it in it's primary key representation instead of it's name. The parameter selector also limits me to being able to select only one parameter and does not appear to let me type in commas either. 

I suspect the issue lies in how the variables are stored and retrieved with the new changes that were implemented. I have temporarily downgraded my version to 1.4.5 (commit: 07002274a3a9830c87627cfe3dbb27b555e8abed) and everything was back to the behavior I expected. Although, I would like to ensure I am running the latest version when this breaking issue has been addressed.

afkiwers, 2025-09-25:
@MrARM I just checked and it does not happen with the latest version of the plugin. What version are you running?

`http://localhost:5173/web/part/2/parameters
`
[image]

`http://localhost:8000/plugin/kicad-library-plugin/v1/parts/2.json`

[image]

afkiwers, 2025-10-01:
@MrARM, I will close this issue for now, as it seems this is no longer an issue. Feel free to reopen this one if needed.